basking-in-the-sun2000/solar-logger #2: Some bugs/notes

bdkacz, 2020-04-19:
1) In solcast\_init_.py there are API key type by  hand" not  getting  from config.py.

2) In my.py lines 

 y_exp = 0
 y_tot = 0

should be commented if somebody don't have external  huawei meter, bacause these  values will be always 0 without meter and generate exception

3) In one of files (I don't remember what exactly, because I fixed this few days ago ) there are problems with  indentation  of few lines which make program to crash 

4) Will be nice to post somewhere in readme/manual list of possibles  timezones for time_zone in config.py for example https://stackoverflow.com/questions/13866926/is-there-a-list-of-pytz-timezones

5) In grafana  for daily production there are references for table Huawei_daily which doesn't exist,but  the solution is so simple - just replace Huawei_daily by Huawei :)

But generally it is a great software and thank You very much for it :)

basking-in-the-sun2000, 2020-06-23:
>     1. In solcast_init_.py there are API key type by  hand" not  getting  from config.py.
> 
>     2. In my.py lines
> 
> 
> y_exp = 0
> y_tot = 0
> 
> should be commented if somebody don't have external huawei meter, bacause these values will be always 0 without meter and generate exception
> 
>     1. In one of files (I don't remember what exactly, because I fixed this few days ago ) there are problems with  indentation  of few lines which make program to crash
> 
>     2. Will be nice to post somewhere in readme/manual list of possibles  timezones for time_zone in config.py for example https://stackoverflow.com/questions/13866926/is-there-a-list-of-pytz-timezones
> 
>     3. In grafana  for daily production there are references for table Huawei_daily which doesn't exist,but  the solution is so simple - just replace Huawei_daily by Huawei :)
> 
> 
> But generally it is a great software and thank You very much for it :)

Thanks for the input. Working on most of your comments, but #2. This one might need more work, since it assumes you are using the meter in several points, and not sure how changes would impact the rest

5. Huawei_daily is in the long term database (basically a daily summary). So you probably need to connect to that db as well

Glad you found it useful. Should have some changes in a few days. Want to make sure the api key is working first

bdkacz, 2020-07-04:
Few bugs you already fixed since my post (19 april), so for example problem in 3) doesn't  exist now :) 

Generally I love it, especially when Huawei servers (cloud)  died or not showing actual/some data :)

basking-in-the-sun2000, 2020-07-04:
@bdkacz  just added a new commit (5d79eba1dee1b861cbfebb57e72e48f5b07cbcf3) that should consider your comments.  Hope it works for you . Sorry for the delay

bdkacz, 2020-07-04:
Ok I will try. Here is my modification for supla  (new def  of supla, improt supla api and some correction in def write_influx  and def main() - maybe you can add it to your  code as option for other users of supla mew 01 ?
[supla.zip](https://github.com/basking-in-the-sun2000/solar-logger/files/4872631/supla.zip)

bdkacz, 2020-07-04:
Oh  and I have Huawei sun2000-6ktl-m0 without optimizers so I must comment in Huawei.py lines with Optim(..) to prevent software crash :)

basking-in-the-sun2000, 2020-07-05:
> Oh and one more note - I have Huawei sun2000-6ktl-m0 without optimizers so I must comment in Huawei.py lines with Optim(..) to prevent software crash :) Don;t know how programically fix this - you know detect model and skip unused registers

It crashes? Unless they changed the register map, it shouldn't matter. Those registers are in the inverter itself. Would had thought it would return 0 for the values for those fields. I will need to check the docs again, but shouldn't matter if you don't have any optimizers. Will try to look it up in their documentation

I included the supla api. Didn't play with it, so just pasted it. If you don't set the api key, it shouldn't do anything

bdkacz, 2020-07-05:
Yes,it crashed, I don't remember exact error but until I don't comment these lines it unable to start.

Ok I'm waiting for commit with supla and start testing :)

basking-in-the-sun2000, 2020-07-06:
Wouldn't had expected that. I checked the manual and it only states **Signals marked with * are supported only by certain models or standard codes.**

I have called a lot of non existent registers while trying to figure out the register maps, but at most got a timeout. If you can let me know what is going on after the next commit, it would be great

bdkacz, 2020-07-07:
Hi one more question in readme we have   "You should have the log file at"
 and no path - where can I find the log ?

basking-in-the-sun2000, 2020-07-09:
added supla support on ac2965c6ef590db775248c93b130d67114be56ee

Let me know how that works for you

bdkacz, 2020-07-09:
So from beggining

When I leave in config.py empty ibverter_ip i get 3x "--> resend" and finally "unable find inverter" - my ifconfig in attachment - eth0 is my cable connection with rest of home network

After comment  in main.py lines 385-387 it starts

Logs from few minutes of working also in attachment - errors about optimizers

BTW I add print(supla) in loop  from x in fazy   and there are no supla in logs so I'm not sure if it enter into supla()
[ifconfig.txt](https://github.com/basking-in-the-sun2000/solar-logger/files/4898831/ifconfig.txt)
[log.txt](https://github.com/basking-in-the-sun2000/solar-logger/files/4898832/log.txt)

basking-in-the-sun2000, 2020-07-13:
it wasn't really crashing, but getting into an infinite loop. Weird that Huawei doesn't return 0 for the optimizer data if none is present. Just set has_optim to false in the config file. 

If it can't find the inverter (do you have ufw on your system?), it will default to the one in the config file now. 

Also, if you don't have a meter, it should now write the daily data(🤞), which wasn't getting created for you before. Eventually you should see a graph like this one. You might have to run fill_daily_blanks.py to fill all the daily values

[image]

bdkacz, 2020-07-13:
When starting as a service in log I get top error: [Errno 98] Address already in use  - the same when I start manually  by  python 3 main.py only running by "sudo python3 main.py" start correctly

bdkacz, 2020-07-13:
I don't have ofw - it is default setup of  Raspberry Pi OS  for raspberry pi 4

Except "fill_blanks error: 'P_Exp'" in log everything else loks working fine,but I will let you know after day of production (actualltyy there is night in Poland :) )

basking-in-the-sun2000, 2020-07-14:
duh, I thought the problem was the continue, but it is calling a non existent variable (P_Exp). 

Think this should fix it. Will be pushing the change later on, but if you want to try it, I'm attaching a file with the changes

[utils.py.zip](https://github.com/basking-in-the-sun2000/solar-logger/files/4915882/utils.py.zip)

closed 2020-08-20 by basking-in-the-sun2000

WMP, 2021-07-09:
Hello, in new supla firmware is option to write to MQTT, without supla cloud. I rewrite this MQTT with telegraf to influxdb. Is possible to use this data from influxdb or MQTT?

My influx structure, now without PV production:
```
time                host topic                                                                                    value
----                ---- -----                                                                                    -----
1625661570007799345 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/1/power_apparent                1125.59068
1625661570106633657 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/1/power_factor                  0.994
1625661570206593461 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/1/phase_angle                   -4
1625661570306543691 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/total_forward_active_energy   2.64154
1625661570607001337 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/total_reverse_active_energy   0
1625661570706875995 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/total_forward_reactive_energy 0.52072
1625661570807715796 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/total_reverse_reactive_energy 1.182
1625661570910451491 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/frequency                     49.99
1625661571006485928 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/voltage                       231.82
1625661571106702367 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/current                       0.787
1625661571207056416 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/power_active                  86.94076
1625661571307314952 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/power_reactive                -48.1335
1625661571607461749 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/power_apparent                179.00546
1625661571706436154 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/power_factor                  0.476
1625661571806657300 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/2/phase_angle                   -24.4
1625661571906430376 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/total_forward_active_energy   6.05874
1625661572006526412 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/total_reverse_active_energy   0
1625661572107015828 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/total_forward_reactive_energy 0.50094
1625661572206664833 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/total_reverse_reactive_energy 1.7316
1625661572306455872 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/frequency                     49.99
1625661572607240642 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/voltage                       231.49
1625661572706656185 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/current                       8.644
1625661572806838220 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/power_active                  1954.14274
1625661572906780753 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/power_reactive                -75.94048
1625661573008225652 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/power_apparent                1961.87404
1625661573107051920 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/power_factor                  0.995
1625661573206118272 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/phases/3/phase_angle                   -2.2
1625661573703539337 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/support                                65535
1625661573802073805 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/total_forward_active_energy            29.11634
1625661573903341015 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/total_reverse_active_energy            0
1625661574002848795 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/total_forward_active_energy_balanced   29.1253
1625661574101985886 ha   supla/devices/zamel-mew-01-3a23c/channels/0/state/total_reverse_active_energy_balanced   0

```

MQTT structure:
[image]

ScApi, 2022-03-03:
I'm trying to properly configure Supla part but no matter what I try most of the times i get
`supla error: Invalid base64-encoded string: number of data characters (5) cannot be 1 more than a multiple of 4`
It's probably something trivial but I'm missing it.