basking-in-the-sun2000/solar-logger #5: Error Starting service

MaxPomy, 2020-09-26:
Good afternoon,

Could you help me?

When I start the service I find the following error

statring setup
opening modbus
opening db
setting db
setting params
main 1 error: list index out of range
main 2 error: list index out of range
SELECT min("Insulation") as "Insulation" FROM "Huawei_info" WHERE time >= 1600992000000000000 and time <= 1601078400000000000 and Insulation > 0

I don't know how to solve it.

Thank you very much.

basking-in-the-sun2000, 2020-09-27:
guess you are doing a new install? The system should be able to go beyond those errors, for they let you know you couldn't retrieve data from the db (the first one could also be you don't have a meter).

Does it continue running, or it crashes? If it runs, it should be ok after a few days of gathering data. It needs to be up at midnight to record some of these values

MaxPomy, 2020-09-28:
Good afternoon,

I've already managed to make it work, I keep seeing those entries in the log traces.

But now I can visualize with graphs the data that is being taken from the inverter and transferred to the influx.

I am very grateful for the effort made to carry out this project.

Best regards

basking-in-the-sun2000, 2020-09-30:
@MaxPomy,

Glad that it is working for you now. If you could let me know what it was, would appreciate it.

Thanks :)

MaxPomy, 2020-10-01:
Good afternoon,

Of course the problem was that I couldn't create the logger table on the influx. I understand this would be due to lack of permissions. I gave him GRANT ALL PRIVILEGES and he started to display the data.
What is interesting is the possibility of entering the value of the KWH for import and the KWH for export. This way we can calculate the balance and know how much money will be sold on the invoice.

Thank you very much!

basking-in-the-sun2000, 2020-10-01:
Glad it is working and thanks for the info. Added info on how to grant privileges on 7710495e1cf32e192e68cc05fc25895912273931. 

Thanks

closed 2020-10-01 by basking-in-the-sun2000