basking-in-the-sun2000/solar-logger #7: Suggestion re Modbus and Registers

craigcurtin-dev, 2020-10-24:
This looks great as previously discussed. I have 2 x Solis inverters and intend to try and adapt this to them. However looking at the bigger picture and what you have tried to achieve with seperating out the register map into its own file i wonder if that could not be taken a step farther and also move the retrieval of those values into a different manner.

In my case my inverters are access by RS485 - Modbus/RTU - i am sure others would have Modbus over Serial/RS232 etc.

I wonder if there is not a way to make a generic "middleware" that your system would use to populate the values, whilst seperate fetchers would be in  charge of communicating with energy meters and other inverters etc.

Happy to work with you (i am not a developer but can bumble my way around in python, C++ and Javascript.) If you are interested.

Craig

basking-in-the-sun2000, 2020-10-24:
Thanks

Currently, it is using Modbus/TCP. However, the library (pythonmodbus) should provide support for rtu and serial connections. 

To provide support for the other methods, I could imagine adding a config flag to indicate the method. However, have no way to test it, so I would be cautious to do this. If you could test other connection methods, I could try to look at adding these options. Not sure if it ought to be a general choice or per inverter. 

Appreciate your offering of help, and you can help me a lot by debugging these alternate methods.

craigcurtin-dev, 2020-10-24:
Yep its hard to test when you only have a single inverter.

I will start messing with this and see if i can get the outline working with my inverters over RS485, then revert back to you and see how we can integrate it further

Craig

basking-in-the-sun2000, 2020-12-27:
Craig, 

Closing this up. If you have any updates, let me know.

closed 2020-12-27 by basking-in-the-sun2000