basking-in-the-sun2000/solar-logger #9: Logging fails if solfor is disabled in config.

javierin, 2021-01-14:
Hi there

I've been using your scanner script to do monitoring of the inverter on my side but i don't want to reinvent the wheel for graphing inverter status with all the amazing work you have done with grafana yourself. 

I believe i have hit a bug or something not considered at coding time. If I run main.py with solfor disabled, I run into several errors regarding missing forcast measurement in logger database. This prevents any data to be recorded into influxdb.

When I run main.py i get the following errors:
```
Thu Jan 14 08:46:20 2021 PF
map looping
read register error Model 1
{'Model': 'SUN2000-5KTL-L1', 'SN': 'HV206004xxxx', 'strings': 2, 'Pn': 5.0, 'Pmax': 5.0, 'Smax': 5.0, 'Qmax': 3.0, 'Qgrid': -3.0, 'Insulation': 3.0, 'Start': '2021-01-14 08:29:54', 'Shutdown': '2021-01-13 18:21:42', 'Time': '2021-01-14 09:46:44', 'Optim_tot': 14, 'Optim_on': 14, 'Optim_opt': 5006}
0
164.55
0
1611.71
0
0.56
0.941
2021-01-14 08:29:54
2021-01-13 18:21:42
20.1
3.0
{'State1': 'grid-connected | grid-connected normally', 'State2': 'unlocked | connected | DSP collecting', 'State3': 'on-grid | off-grid switch disable', 'Alarm1': '', 'Alarm2': '', 'Alarm3': '', 'Status': 'On-grid (Off-grid mode: running)', 'Fault': '0'}
looking at low production
{'PV_Un': 264.8, 'PV_In': 1.58, 'PV_P': 0.805, 'U_A-B': 224.9, 'U_A': 202.7, 'U_B': 0.3, 'I_A': 3.519, 'P_peak': 0.941, 'P_active': 0.786, 'P_reactive': 0.001, 'PF': 1.0, 'Frequency': 49.99, 'η': 97.56, 'Temp': 20.1, 'P_accum': 495.14, 'P_daily': 0.56, 'M_P': -741, 'M_Pr': 428, 'M_A-U': 225.3, 'M_B-U': 0.0, 'M_C-U': 0.0, 'M_A-I': -3.78, 'M_B-I': 0.0, 'M_C-I': 0.0, 'M_PF': -0.864, 'M_Freq': 49.99, 'M_PExp': 164.55, 'M_U_AB': 0.0, 'M_U_BC': 0.0, 'M_U_CA': 0.0, 'M_A-P': 0, 'M_B-P': 0, 'M_C-P': 0, 'M_PTot': 1611.71}
1610614259.3561013
top error: name 'forecast_array' is not defined
something went wrong

inverter address: 192.168.1.30
192.168.1.30
192.168.1.30
statring setup
opening modbus
opening db
setting db
setting params
main 1 error: list index out of range
main 2 error: list index out of range
SELECT min("Insulation") as "Insulation" FROM "Huawei_info" WHERE time >= 1610496000000000000 and time <= 1610582400000000000 and Insulation > 0
starting
in forecast
```

Influxdb contents:
```
> use logger
Using database logger
> show measurements
name: measurements
name
----
Huawei_info
Huawei_stat
> select * from Huawei_stat;
name: Huawei_stat
time                Fault State1                                   State2                                State3                            Status                           location
----                ----- ------                                   ------                                ------                            ------                           --------
1610613098922815540 0     grid-connected | grid-connected normally unlocked | connected | DSP collecting on-grid | off-grid switch disable On-grid (Off-grid mode: running) Casa
1610613165560451854 0     grid-connected | grid-connected normally unlocked | connected | DSP collecting on-grid | off-grid switch disable On-grid (Off-grid mode: running) Casa
1610613339139643541 0     grid-connected | grid-connected normally unlocked | connected | DSP collecting on-grid | off-grid switch disable On-grid (Off-grid mode: running) Casa

> select * from Huawei_info;
name: Huawei_info
time                Insulation Model           Optim_on Optim_opt Optim_tot Pmax Pn Qgrid Qmax SN           Shutdown            Smax Start               Time                location strings
----                ---------- -----           -------- --------- --------- ---- -- ----- ---- --           --------            ---- -----               ----                -------- -------
1610613098908009845 3          SUN2000-5KTL-L1 14       5006      14        5    5  -3    3    HV206004xxxx 2021-01-13 18:21:42 5    2021-01-14 08:29:54 2021-01-14 09:31:40 Casa     2
1610613165540046855 3          SUN2000-5KTL-L1 14       5006      14        5    5  -3    3    HV206004xxxx 2021-01-13 18:21:42 5    2021-01-14 08:29:54 2021-01-14 09:32:45 Casa     2
```

As soon as solfor is enabled, forcast measurement is created and everything works smooth. 

Kudos!

basking-in-the-sun2000, 2021-01-21:
Sorry for the delay. I think I know what is going on, and will take a look

basking-in-the-sun2000, 2021-01-23:
think the new changes should fix your problem

basking-in-the-sun2000, 2021-01-23:
@javierin any reason you don't want to use solcast?

javierin, 2021-01-23:
> @javierin any reason you don't want to use solcast?
> 
> Also the low production emails only work from 11 to 15 h when production drops to 0

I didnt have the api key around when i initially set it up and found this behavior.

basking-in-the-sun2000, 2021-01-23:
thought there was some other reason (privacy, remote access, etc.).

The new update should fix this 🤞. Thanks for giving the logger a go :)

closed 2021-01-23 by basking-in-the-sun2000