basking-in-the-sun2000/solar-logger #11: Hassio - Home Assitant integration

johanvh83, 2021-01-22:
Hi, is this an integration into Hassio (Home Assitant)?
I plan to buy/install a solar installation with this specific converter and an additonal battery, so basically at least the solar panel logging should work.

basking-in-the-sun2000, 2021-01-23:
Never looked into integrating it with other systems. Basically I thought of it as a self contained unit

Also don't have a battery, so didn't add that. We could work on it to include battery info

basking-in-the-sun2000, 2021-06-26:
If you could provide some feedback could take a look, don't want to add the battery info since I have no way to test it.

jhoms, 2021-06-27:
I have an LG Battery attached and I managed to find 2 addresses that work:

```
    'S_C_n_D':    {'addr': '37001', 'registers': 2,  'name': 'Storage charge and discharge',     'scale': 1,    'type': 'I32',  'units': 'W'   , 'use': 'data',  'method': 'hold'},  
    'S_SOC':      {'addr': '37004', 'registers': 1,  'name': 'Storage_State Of Charge',          'scale': 10,   'type': 'U16',  'units': '%'   , 'use': 'data' , 'method': 'hold'},  
```

basking-in-the-sun2000, 2021-06-28:
Thanks. Will be adding a battery related section to the registers. Would guess there are a bit more, but maybe one day :)

My guess is that there are some at 37002-3, 37006-14

Will try to push the changes soon

if (config.has_ESU) :
    _register_map.update({
        'ESU_status':  {'addr': '37000', 'registers': 1,  'name': 'ESU status',                       'scale': 1,    'type': 'U16',  'units': ''  ,   'use': 'status',  'method': 'hold'},  
        'ESU_power':   {'addr': '37001', 'registers': 2,  'name': 'ESU power',                        'scale': 1,    'type': 'I32',  'units': 'W'   , 'use': 'data',  'method': 'hold'},  
        'ESU_soc':     {'addr': '37004', 'registers': 2,  'name': 'ESU SOC',                          'scale': 10,   'type': 'U16',  'units': '%'   , 'use': 'data',  'method': 'hold'},
        'ESU_charge':  {'addr': '37015', 'registers': 2,  'name': 'ESU Current Charge',               'scale': 100,  'type': 'U32',  'units': 'kWh' , 'use': 'data',  'method': 'hold'},
        'ESU_discharge':{'addr': '37017','registers': 2,  'name': 'ESU Current Discharge',            'scale': 100,  'type': 'U32',  'units': 'kWh' , 'use': 'data',  'method': 'hold'}})

basking-in-the-sun2000, 2021-07-16:
should be included in the current version (https://github.com/basking-in-the-sun2000/solar-logger/commit/5af778af5cf01908f954668fb8213202f7b75f40)

closed 2021-07-16 by basking-in-the-sun2000