basking-in-the-sun2000/solar-logger #12: Smartlogger - more than one inverter

bdkacz, 2021-01-22:
Hello.

I have Smartlogger 3000A and two SUN2000-60KTL-M0. After create  Smartlogger.py and set registers from smartlogger modbus manual I can get values of smartlogger (cummulative production of both inverters etc) but I dont't know how to get data from single inverter (for example voltage and current of each  string). Webui of smartlogger display this data, so smartlogger itself  have access to it . Modbus manual  of smartlogger dont't explain enaught easy how to do it.

basking-in-the-sun2000, 2021-01-23:
I originally got a smartlogger 1000 since the tech person mentioned it should work for my needs. It didn't :( So that is why I made the logger. However, will have to think how to add more inverters to the logger. 

Are you connecting through the modbus or tcp? Also can you access the inverter data through the smartlogger? The reason is I thought you could only address the inverter through the modbus address 0 when using tcp. Might be possible to access each with their own IP address, but might be easier if you can read the logger

bdkacz, 2021-01-23:
via modbus tcp (I enable it on smartlogger)   - when I change the  slave address in config.py to 0x01 or 0x02 I can normally read data from the inverters, but only from one at the same time

In documentation of modbus for smartlogger I dopn't find info how to access to data from each inverter via smartlogger address (I can only read total production of all inverters etc)

Yes I have admin access to the smartlogger

bdkacz, 2021-01-23:
One more - because the smartlogger don't send insulation info there is much error about it when I try to get data from smartlogger

basking-in-the-sun2000, 2021-01-23:
let me think about how to work this out. Didn't consider power user's problems when I designed it lol Still trying to find the smart logger register map, but should be able to do something about reading 2 inverters.

Never considered that you couldn't read the insulation resistance. I'm tracking some values to warn you of issues during the lifetime and consider these vital. These include the insulation, wiring resistance (though this is only displayed on the dashboard), etc. Recently found a breaker heating up. After reseting it a few times in hopes of improving the connection, the wiring resistance dropped. Also gave the screws along the main wiring a once over, since some had loosen up a bit after a year.

bdkacz, 2021-01-23:
SmartLogger Modbus documentation https://support.huawei.com/enterprise/en/doc/EDOC1100050690 on page 29 (of file)  defines way to access data from each inverter bby  remaping addresses but there is only few basic information - not all

bdkacz, 2021-01-23:
Yes I know that You create it for your home use , but fusion solar app working horrible from day to day , dediacted huawei "local" solutions cost much money  and I have good experiences with your software with my private installation. So will be nice if I can use it also in my company :)

bdkacz, 2021-01-23:
Reading data from smartlogger is not necessary from me - more improtant is possible of reading from two  SUN20000  at the same time  - because when I will have data from alll devices I can "calculate" smnartlogger data in grafana :) 

BTW in  measure Huawei  IMHO we need to add  SN of inverter, or other "id" to  separate data frokm each inverter  :)

I try to adapt the software by myself (I'm a CA# programmer), but in code of main.py  there is a lot of code strictly from readings mixed with the code from the forecast :)

basking-in-the-sun2000, 2021-01-23:
lol I can't log onto the Huawei repository (never got a email confirmation with a code), so your link helped. Thanks.

I saw, but ignored it, that it was for a commercial inverter (SUN2000-60KTL-M0). The register maps are different for commercial and residential. They have Interface Definitions V2.0 vs V3.0. The included Huawei.py might not work for your inverters.

Will take a look at the smart logger, but Huawei works from the idea of security through obscurity, since they don't publish all the info. You could try the scanner files to see if you find a more complete map. That is partly how I managed to provide more info than what Huawei publishes in their Interface Definitions.

Will take me a bit of time to figure the 2 inverter issues, so have patience. There is a parameter that might help identify each, but the grafana interface never considered this. So it might turn into a major hassle. The parameter is the location value in the config file. Not sure how messy reading will become.

Also glad you think the inverter is good enough for work :) I'm surprised that the smart logger didn't solve your needs. I thought it did what the solar-logger did and a lot more (they have a nice pf adjustment based on the meter value). Hope we can work out something you can use

bdkacz, 2021-01-23:
SmartLogger may have a nice web interface, but the only thing you can do with the data is to export to a csv file and what Huawei found "useful information" ... so it's not the best solution :)

Besides, I don't really know how much data a smartlogger can hold (what memory it has), and here I am limited only by the disk of the machine on which I will run your logger :)

Creating a map for the SUN2000-60 is, I think, something that I can handle on my own, worse with the reading :)

Hope you would like this "challenge" :)

bdkacz, 2021-01-23:
Here is modbus definition for SUN2000-60 and other "big" inverters https://drive.google.com/file/d/1FUepL_27UJQedIEemEEi54-V4V0RdADE/view?usp=sharing

bdkacz, 2021-01-23:
BTW why You  choose influxdb, not for example MySQL ?

basking-in-the-sun2000, 2021-01-23:
influxdb unlike sql db should be better designed for time series. It has a lot of sql commands, but also has a lot of limitations, though it should be better designed for the type of data you get from the inverter. The new flux language should overcome some of the limits. I have managed most of the garfada calls using the pseudo sql calls. 

https://en.wikipedia.org/wiki/Time_series_database
https://en.wikipedia.org/wiki/InfluxDB

bdkacz, 2021-01-23:
I see influxdb 2.0 but if I good see it has another connection string etc so migrating is  probably not easy :)

bdkacz, 2021-01-23:
Personally, in influxdb I am annoyed by operating conditions like -12h etc. There is no simple way to query for data, e.g. from midnight to now, or simply to query for a date range from to

basking-in-the-sun2000, 2021-01-24:
indeed it has a lot of limitations. Time is one, some queries are either very difficult or almost impossible. The new flux helps, but not perfect. 

For time either it helped doing it in python, or garfada has ways around it. 

Even with all that, in theory, time series should be better at handling large amounts of sequential data. Maybe influxdb wasn't the best choice, but seems solid in general

basking-in-the-sun2000, 2021-01-24:
@bdkacz The smart logger doesn't provide all the registers, but you can access the most common through the 2.7 Remapped Modbus definitions. The meter is access through 2.4 Register Definitions for the Power Meter. The one that doesn't seem too clear is the 2.5 Register Definitions for the SUN2000. 

Regardless, I will take a look at allowing more inverters to be polled by the solar-logger

bdkacz, 2021-01-24:
Question abount   each string voltage and current. In your Huawei.py I found   'PV_Un' and  'PV_In': reffers to PV1 string. What about rest of strings ? I must manually add each ?

bdkacz, 2021-01-24:
Solution for strings. In Huawei.py add

for x in range(config.strings):
    _register_map.update({
        'PV'+str(x+1) +'V' :   {'addr': '320'+str(16+2*x), 'registers': 1,  'name': 'PV ' +str( x) +' voltage',      'scale': 10,    'type': 'U16',  'units': 'V'    , 'use': 'data',  'method': 'hold'},
        'PV'+str(x+1) +'A':  {'addr': '320'+str(17+2*x), 'registers': 1,  'name': 'PV ' + str(x) + ' current',           'scale': 100,    'type': 'U16',  'units': 'A'    , 'use': 'data',  'method': 'hold'}})

and in config.py  attribute strings with number of strings in inverter

basking-in-the-sun2000, 2021-01-24:
I started with that idea, and don't recall why I decided not to go with it. Adding a loop for the extra strings doesn't seem complicated, and the docs mention you could have up to 24 strings.

Maybe thought that most residential units would only have one string (told you, power user's problems lol)? Could have been a problem with displaying all the strings (maybe it was to make it aware to the user, that they would need to add the extra strings to the graphs). Honestly, don't recall, but there was a reason why I decide against.

bdkacz, 2021-01-24:
My home sun200p 6kW  have 2 strings :)

bdkacz, 2021-01-24:
In atttachent mapped file for Huawei "big" inverters - based on Huawei.py and corrections from documentation for enterprise devices[
[HuaweiEnt.zip](https://github.com/basking-in-the-sun2000/solar-logger/files/5863058/HuaweiEnt.zip)

](url)

bdkacz, 2021-01-30:
Hi :) any news ? :)

basking-in-the-sun2000, 2021-02-07:
just posted an upgrade for multiple strings 185b70fb32fe247ef2b8295724d4290509deb4bf

bdkacz, 2021-02-07:
Thanks :) So now just waiting for 2+ inverters support ;)

bdkacz, 2021-03-16:
Hi :) Any progress with more than one inverter ? :)

basking-in-the-sun2000, 2021-05-16:
been giving this some thought and not sure I want to open that can of worms. Assuming you are using residential units and connecting using Modbus TCP, you would want to have workers connecting to the inverters. Otherwise the connection time could increase if you do it sequentially. This would be a major rewrite, that don't have the time for at the moment. For commercial units or using modbus RTU, you don't want to do this, since modbus has no collision safeguards. 

Unless there is a brave soul wanting to tackle this, I'm going to close it for now

closed 2021-05-16 by basking-in-the-sun2000