basking-in-the-sun2000/solar-logger #13: 'forecast_array' is not defined

bdkacz, 2021-01-23:
Hi

When there is problem with getting forecast we get  "'forecast_array' is not defined" exception, probably because in line 487 of main.py we have    for loop but  without checking if it is not null

basking-in-the-sun2000, 2021-01-23:
just added that for the issue that javierin posted. Hope the current push will fix your problems. Just as a warning, the low production emails only work from 11 to 15 hours when the production drops to 0. 

Any reason you don't want to use the solecast forecasts?

bdkacz, 2021-01-23:
For example now (I don't know how)  I exceed the 20 cal;ls/day  and...solar-logger give error and  don't write data from inverter to database :) So it is not "I don't want to use" but for example  solcast is down for maintance and we lost logging function :)

basking-in-the-sun2000, 2021-01-23:
if you restarted the application multiple times, it can go over the limit. Most of the time I'm around 7 calls a day, but do go to 20 occasionally(!). Hope the patch will fix this for you. 

I was curious if someone didn't wanted to use it and for what reason.  Thanks for giving the logger a go :)

closed 2021-01-23 by basking-in-the-sun2000

bdkacz, 2021-01-23:
Las update solve the problem- thanks ;)