basking-in-the-sun2000/solar-logger #14: Connecting Pi to the Inverter

heyeolasad, 2021-01-26:
First thank you for your code and contributions. It's an amazing project. I have a question.

It may seem like trivial but how do I connect the Pi to the inverter? I'm searching for this but I have no idea.

bdkacz, 2021-01-26:
Via  wi-fi. You must connect  pi by wire (or second usb  wi-fi adapter)  to your home network and via w-fi with network named SUN2000....

pepoo82, 2021-01-26:
It also works if the inverter is connected to your local network via wifi, you can get the data from your LAN to any device.

basking-in-the-sun2000, 2021-01-27:
Originally planned on connecting the rpi to the modbus cabling. Thought of having the rpi work as a repeater between the inverter and the rest of the bus. However, the wifi has worked great, and reduced the risk of a surge. I'm sure Huawei isolated the rs485 well, but if there is a lighting, it might damage more than you would like. So as pepoo and bdkacz said, you can use the wifi (either the builtin AP or your network). 

Mine is on a separate lot wifi network, and I can access the grafana graphs by using that network. If you have the rpi on the inverter's AP network, not sure you would be able to access the rpi (though never tried it) from your computer. You might then have to have the rpi on 2 networks (wifi and ethernet cable)?

basking-in-the-sun2000, 2021-06-26:
hope the new changes helped (https://github.com/basking-in-the-sun2000/solar-logger/commit/2c7569bdd2596276fb03ed6497876ad78d565e58)

WMP, 2021-07-11:
I connect to SUN2000-xxxxxx WiFi by place nearly cheap router with Openwrt, connect to my network, connect to SUN2000-xxxxxx WiFi with password Changeme as client, set static IP (ex. 192.168.200.99/24), and then i set static route to 192.168.200.0/24 over my cheap router IP, and i set MASQUERADE on this router to from my LAN to SUN2000 network.

basking-in-the-sun2000, 2021-07-11:
@WMP thanks great share. I added it to the discussion section

https://github.com/basking-in-the-sun2000/solar-logger/discussions/24

closed 2021-07-11 by basking-in-the-sun2000

verisgit, 2022-04-22:
Any RPI with Ethernet and Wifi can connect to the Huawei built-in AP and your LAN at the same time. i.e. you can SSH to the RPI over ethernet on 192.168.1.xxx for example whilst at the same time the RPI is connected to the Sun2000 Wifi communicating with the Inverter at 192.168.200.1.

Note they must be on different subnets. The L1 seems to come with 192.168.200.1 as the default but you can change it in the Sun2000 app by accessing the inverter in installer mode if for some reason your home network has the same subnet.

I've tested this today and it seems to be working fine.

Alternatively, you can buy an SDongle-5A and that will apparently provide access to the Modbus ports on your LAN. The Sun2000-5KTL-L1 doesn't seem to allow access to Modbus ports from the LAN - only from it's internal network. Apparently, the SDongle bypasses this issue (and opens a bunch of security holes if you aren't careful) - but I'm yet to get my hands on one to test it out.