basking-in-the-sun2000/solar-logger #15: Download of optimizer data

Paco93, 2021-02-09:
Hello,
thanks for the great Software you provided us. I have a SUN2000-3KTL with smart PV optimizers. I was wandering how the data from the optimizer could be downloaded and made available.
Regards

basking-in-the-sun2000, 2021-02-14:
Huawei doesn't give all the details on how to download the optimizer info. They said it is proprietary. Their modbus reference manual, gives some hints, but not enough info

closed 2021-04-30 by basking-in-the-sun2000

madimadi, 2022-02-18:
Hi. I believe there is a section 4.2 in interface definition document 2021-08-13 about real-time data query of optimizers.

Would you consider adding support for it?
Regards,
 madimadi

andysu, 2022-02-20:
not so,
some basic info you may read from registers (only from m3 inverters)
- Total number of optimizers
- Optimizers Online
- Optimizer Feature data
additional info (only thru fusion solar and 3rd app)
--Basic
  - SN
  - address
  - string N
  - pos in string
  - FW ver
--Realtime
  - Alarm
  - Input current, voltage
  -Output current, voltage
  -temperature
  -status
  -Power
  -voltage to ground
  -accumulated energy yield

madimadi, 2022-02-20:
Sorry I'm not sure still.  You say real-time data cannot be read only basic and basic can be read only in M3. Not in smaller inverters ex: M1?

andysu, 2022-02-20:
sorry - bad markup in my post
fix.
M1 inverter - please send model and FW info

basking-in-the-sun2000, 2022-02-20:
Some inverters do not use optimizers. Still, it would be great, but section  4.2(?) doesn't provide enough info to download the optimizer info. It is really a bad move from Huawei not allowing us to access that info :<

Have the impression you need to login to the inverter to access that. The remote monitors don't need it, since the inverter pushes the data out.

andysu, 2022-02-21:
imho 
some data can be downloaded (func 0x44 and 0x41 is the same) without auth process
I did not find any information that it is possible to organize the authentication process through modbus
I have not yet been able to download data from the inverter. But I'm trying.
I was able to decipher some EMAP files. without huawei info
In general, the topic is very interesting. So you can download a lot of information about the operation of the inverter.
I propose to combine efforts.

andysu, 2022-02-21:
I read your correspondence on the forum Hua

basking-in-the-sun2000, 2022-02-25:
sounds interesting. If I can help, let me know

andysu, 2022-02-28:
Thanks, friend. After war. I'm in Ukraine

basking-in-the-sun2000, 2022-02-28:
Stay safe and hope this madness is done soon

andysu, 2023-06-09:
some updated info
Huawei added information about optimizers
**4.1 Obtaining the System Information of Optimizers**
according to this information, I was able to write with an application that reads optimizers (live and history)

basking-in-the-sun2000, 2023-06-09:
> some updated info Huawei added information about optimizers **4.1 Obtaining the System Information of Optimizers** according to this information, I was able to write with an application that reads optimizers (live and history)

Excellent!. Would you mind sharing the updated documents and if you have sample code.

andysu, 2023-06-09:
Updated documents - OK (pls send me your email address for file)
sample code - not right now (its my commercial property)
maybe later.

basking-in-the-sun2000, 2023-06-09:
> Updated documents - OK (pls send me your email address for file) sample code - not right now (its my commercial property) maybe later.

No problem. The docs will do.

andysu, 2023-06-09:
the libmodbus standard library has support for custom functions, but it does not work on modbus RTU.