basking-in-the-sun2000/solar-logger #16: How to connect to Inverter?

KlausHans, 2021-04-21:
Hello,
i hate the app from huawei very much, it is so unreliable and produces false information. So i found your nice project. Thanks for sharing.
I have a SUN2000-17 KTL-M0, no smartlogger.
Unfortunately i have troubles to get it working. If i run `python3 main.py` i get a timout:
```
pi@raspberrypi:~/basking-in-the-sun2000/solar-logger $ python3 main.py
mailing error: [Errno -2] Name or service not known
starting
Wed Apr 21 21:18:59 2021
huaweil inv_address error: timed out
huaweil inv_address error: timed out
huaweil inv_address error: timed out
huaweil inv_address error: timed out
```
I got the ip adress (192.168.178.28) from my router, but this is somewhat a good guess. How can i get the ip adress? Of course it is not displayed in the Fusion app.
Also i am wondering: is the port always the same?
Do you have an idea how i can get the connection?

Off topic: as i run your project for the first time, it got errors because the python modules `isodate` and `gpiozero` where missing. You can add this to the readme.

KlausHans, 2021-04-21:
Just read here, that the inverter does not communicate if in standby (=no sun).
https://community.openhab.org/t/reading-data-from-huawei-inverter-sun-2000-3ktl-10ktl-via-modbus-rtu/87670/6

basking-in-the-sun2000, 2021-04-22:
Thanks for the heads up with the issues you had. Tried to make it clearer in the docs. 

Should be included in https://github.com/basking-in-the-sun2000/solar-logger/commit/30972e2447da4379beb51b393ab949cfb2863ba2

closed 2021-04-22 by basking-in-the-sun2000

WMP, 2021-07-11:
Sun2000 m1 works in night, response on  modbus tcp

basking-in-the-sun2000, 2021-07-11:
Thanks, great to know. Had heard some units didn't respond after sundown.