basking-in-the-sun2000/solar-logger #22: ERR: invalid operation: time and *influxql.BinaryExpr are not compatible

WMP, 2021-07-11:
Query: 
```
select cumulative_sum(diff)  as "∑ balance" from (SELECT first("P_Exp") - first("P_Grid") + first("Adj") as diff FROM "Huawei_daily" WHERE time > '${__to:date:YYYY-MM-18}' - 52w  GROUP BY time(1d) fill(0));
```
on influxd 1.8.1-1 (ubuntu 20.04) return this error: `ERR: invalid operation: time and *influxql.BinaryExpr are not compatible`

I think that this same error is describe here: https://discourse.nodered.org/t/value-calculation-last-day/42419

WMP, 2021-07-11:
Upgrade garfana to version 8 fix this problem.

closed 2021-07-11 by WMP

basking-in-the-sun2000, 2021-07-11:
@WMP 
:) glad you got it fixed.  The **${__to:date:YYYY-MM-18}** is internal to grafana and not used by influx. Didn't realize it was a recent change, and you needed version 8.

Actually, this is going to disappear. Turns out that it doesn't show the actual balance after a while. At least on net-metering, it shows the cumulative sum of the past year, but not your current energy balance. The new method should deduce the months you didn't have excess from those you did before. Hopefully will be ready soon.