basking-in-the-sun2000/solar-logger #25: Multiple string setup

lollul, 2021-07-24:
Even though I have specified strings = 2 in config the logger would not read each string voltage/current values.

The issue seems to be in Huawei.py. I've changed the loop processing:
```
--- a/Huawei.py
+++ b/Huawei.py
@@ -82,7 +82,7 @@ _register_map =  {
 }
 
 if (config.strings > 1):
-    for i in range(2, config.strings):
+    for i in range(1, config.strings + 1):
         _register_map.update({
             'PV_U' + str(i):      {'addr': str(32014 + i * 2), 'registers': 1,  'name': 'PV' + str(i) + ' voltage',                      'scale': 10,   'type': 'I16',  'units': 'V'   , 'use': 'mult',  'method': 'hold'},  
             'PV_I' + str(i):      {'addr': str(32015 + i * 2), 'registers': 1,  'name': 'PV' + str(i) + ' current',                      'scale': 100,  'type': 'I16',  'units': 'A'   , 'use': 'mult',  'method': 'hold'}})
```

Before:
{'PV_Un': 351.2, 'PV_In': 11.2, 'PV_P': 7.841, 'U_A-B': 413.1, 'U_A': 239.0, 'U_B': 238.3, 'I_A': 10.65, 'P_peak': 0.0, 'P_active': 7.629, 'P_reactive': 0.003, 'PF': 1.0, 'Frequency': 50.0, 'η': 97.3, 'Temp': 61.0, 'P_accum': 638.95, 'P_daily': 25.4, 'M_P': 0, 'M_Pr': 0, 'M_A-U': 0.0, 'M_B-U': 0.0, 'M_C-U': 0.0, 'M_A-I': 0.0, 'M_B-I': 0.0, 'M_C-I': 0.0, 'M_PF': 0.0, 'M_Freq': 0.0, 'M_PExp': 21474836.47, 'M_U_AB': 0.0, 'M_U_BC': 0.0, 'M_U_CA': 0.0, 'M_A-P': 0, 'M_B-P': 0, 'M_C-P': 0, 'M_PTot': 21474836.47, 'PF_comp': 1.0, 'Q/S': 0.0}
After:
{'PV_Un': 367.1, 'PV_In': 11.2, 'PV_P': 8.197, 'U_A-B': 415.5, 'U_A': 240.4, 'U_B': 240.8, 'I_A': 11.041, 'P_peak': 0.0, 'P_active': 7.977, 'P_reactive': -0.001, 'PF': 1.0, 'Frequency': 50.0, 'η': 97.32, 'Temp': 60.4, 'P_accum': 648.82, 'P_daily': 35.27, 'M_P': 0, 'M_Pr': 0, 'M_A-U': 0.0, 'M_B-U': 0.0, 'M_C-U': 0.0, 'M_A-I': 0.0, 'M_B-I': 0.0, 'M_C-I': 0.0, 'M_PF': 0.0, 'M_Freq': 0.0, 'M_PExp': 21474836.47, 'M_U_AB': 0.0, 'M_U_BC': 0.0, 'M_U_CA': 0.0, 'M_A-P': 0, 'M_B-P': 0, 'M_C-P': 0, 'M_PTot': 21474836.47, 'PF_comp': 1.0, 'Q/S': 0.0, 'PV_U1': 367.4, 'PV_I1': 11.2, 'PV_U2': 367.3, 'PV_I2': 11.2}

BTW Have you considered using: https://gitlab.com/Emilv2/huawei-solar

basking-in-the-sun2000, 2021-07-25:
🙈 Sorry about that. I had done something similar in another language, and limiting the upper range to a value did include that value. In python, it doesn't, for instance, if the limits are (2,2). Anyhow, sorry didn't run it since it looked so trivial, guess it wasn't

However, I did debate for a while how to work that out, and in order to prevent people with 1 string from either having to rename all their series or having discontinuities, I did leave the PV_Un naming. So string 2 should have been PV_U2, and string 1 PV_Un. 

Even though there seems to be a small difference (probably different reading times), PV_Un and PV_U1 should have been the same. Since you are on this boat, would you rather use PV_Un as string 1 or PV_U1. If you prefer the 1, people with 1 string would still get PV_Un.

lollul, 2021-07-25:
Sorry, I've wrongly assumed that PV_Un is some kind of an average reported by inverter. But I see now it is read from the same registers. Your patch then makes more sense certainly. 
Although maybe we could collect both PV_U1 and PV_U2 (etc) while PV_Un would be calculated as average of the two? But not sure if there would be any value in this... I've only connected my installation few days ago and I'm just trying to understand how things work.

basking-in-the-sun2000, 2021-07-25:
I like your idea. Will try to use n as the total for all the strings. This would make it backward compatible with those who have used it for a while. Thanks

basking-in-the-sun2000, 2021-07-27:
Checkout https://github.com/basking-in-the-sun2000/solar-logger/commit/c3217466a7494f39823b467147d175abed160786, it should allow people with multiple strings to include those in the data. Vn gives the average for the strings (might need to consider if there are uneven) and In is the total current

lollul, 2021-07-28:
I've been running c321746 today. I've noticed PV_Un is now a sum of PV_U1 and PV_U2. I was expecting you would use an average for voltage and a sum for amperage...
Example:
{'PV_P': 7.677, 'U_A': 240.0, 'U_B': 237.8, 'U_C': 239.5, 'I_A': 10.443, 'I_B': 10.507, 'I_C': 10.473, 'P_peak': 0.0, 'P_active': 7.47, 'P_reactive': 0.002, 'PF': 1.0, 'Frequency': 50.02, 'η': 97.31, 'Temp': 60.9, 'P_accum': 810.98, 'P_daily': 19.06, 'M_P': 0, 'M_Pr': 0, 'M_A-U': 0.0, 'M_B-U': 0.0, 'M_C-U': 0.0, 'M_A-I': 0.0, 'M_B-I': 0.0, 'M_C-I': 0.0, 'M_PF': 0.0, 'M_Freq': 0.0, 'M_PExp': 21474836.47, 'M_U_AB': 0.0, 'M_U_BC': 0.0, 'M_U_CA': 0.0, 'M_A-P': 0, 'M_B-P': 0, 'M_C-P': 0, 'M_PTot': 21474836.47, 'PF_comp': 1.0, 'Q/S': 0.0, 'PV_Un': 688.5, 'PV_U1': 344.5, 'PV_In': 22.4, 'PV_I1': 11.2, 'PV_U2': 344.0, 'PV_I2': 11.2}

Am I assuming correctly that your inverter is able to measure your local power use? I think I would need a separate smart meter for that.
Also, do you have a two phase system? I mean you declare in Huawei.default.py 'U_C' as 'use': 'ext'

basking-in-the-sun2000, 2021-07-28:
Might be that it prints the debug message before it calculates the average. If the value on the db is correct it is a display error. You can see the stored values on the solar dashboard page. Click on detail (pops up from the link icon, left top) from the Energy panel. 

If the value there is wrong, could change the line (560) for this for one cycle it would help. 

```
            print(measurement["PV_Un"])
            print(config.strings)
            measurement["PV_Un"] = measurement["PV_Un"] / config.strings
            print(measurement["PV_Un"])
```

You do need a meter to get loads and other grid values. Think your system should run without the grid-tied meter (since it pulls it from the inverter itself), but you would have some weird values. Somehow your inverter is stuffing one into M_PExp and M_PTot. Wonder if those are what it considers negatives or NaN? Though one is an unsigned. 

My inverter is only tied between two phases. So you would need to change U_C to _data_ if you have a three phase unit, as you already did.

lollul, 2021-07-29:
Yes, you're right, sorry. All is well with PV_Un.
I'm tempted to start playing with Grafana more, but then I'm worried I would have trouble keeping up with your changes :-)

FYI I'm running the python script on my openwrt router and influxdb and grafana is on a cloud machine. This already works better than the Huawei site. 

Thank you very much for your project!

basking-in-the-sun2000, 2021-07-29:
lol don't worry. I should had placed it above the debug line. Will do it in the next revision

I haven't found a good solution for that. The new version of Grafana allows for a library of panels. So you could import your panels into the general dashboards.

Hmmm. Nice you got it on a router. Never thought of separating the parts, but glad you could.

Thanks

closed 2021-07-29 by basking-in-the-sun2000