basking-in-the-sun2000/solar-logger #27: MQTT Publish

bdkacz, 2021-08-08:
Hi.

Is it possible to add function to publish via MQTT current power and daily production ?

closed 2021-08-08 by bdkacz

basking-in-the-sun2000, 2021-08-08:
It is possible to use mqtt. However since the data is already in influxdb, you could have another process send the data to the broker, or add extra data with Telegraf.

Could you let me know what you are trying to do? Seems you solved this, but curious about your application.

bdkacz, 2021-08-08:
I wanted to do this because after a few hours of trying HA did not want to correctly interpret the influxdb database data. But since I finally succeeded, I closed the issue

WMP, 2021-12-26:
Could you told me how you solve this?

bdkacz, 2021-12-27:
> Could you told me how you solve this?

I create iunfluxdb sensor based on https://www.home-assistant.io/integrations/influxdb/ 

with query

  queries:
    - name: Huawei_Production
      group_function: last
      where: '1=1'
      measurement: '"Huawei"'
      field: P_accum
      database: logger  

and  template sensor using this influxdb sensor   with attributes   
        unit_of_measurement: "kWh"
        device_class: energy
        state_class: total_increasing

f432, 2022-01-07:
Has anyone been able to setup exported energy into Home Assistant using influxdb as a sensor? I've spent days on it but my knowledge is not there,

Thought I would try here before asking in the HA forums.