basking-in-the-sun2000/solar-logger #28: Connection issues

haralddvorak, 2021-09-06:
Hi!
Thanks for this amazing project and all your efforts you've put into it!
I have trouble connecting my inverter and enabled debugging mode in the config file.

My inverter is a **SUN2000-10KTL-M1** with a
**SDongleA-05 SW-Version:V100R001C00SPC123** (latest available version)
which is **connected via LAN**.
I'm using a virtual machine with debian 10, with the latest influxdb and grafana.

I can connect and read a register via a small test script using the inverter IP-adress and the port (IP-Address of inverter removed):
```
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
import time

client = ModbusClient("Edit: IP-Address removed", port=502)
client.connect()

if client.connect():
    time.sleep(1)

    rr = client.read_holding_registers(0x7D50, 0x02, unit=1) #32080 => active_power

    activePowerInv=rr.registers[1]
    print(activePowerInv)
```

Running main.py shows two errors:

```
setting params
main 1 error: list index out of range
main 2 error: list index out of range
```
After fetching the forecast-data the following errors occur:
```
map looping
read register error Model 1
read register error Model 2
read register error Model 3
read register error Model 4
```
Starting scanner.py results in errors trying to read the registers like the one below:
```
DEBUG:pymodbus.transaction:RECV:
DEBUG:pymodbus.framer.socket_framer:Processing:
DEBUG:pymodbus.transaction:Getting transaction 3
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('Edit: IP-Address removed', 54487)
DEBUG:pymodbus.transaction:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.transaction:Running transaction 4
DEBUG:pymodbus.transaction:SEND: 0x0 0x4 0x0 0x0 0x0 0x6 0x0 0x3 0x71 0x49 0x0 0x2
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:No response received to unbounded read !!!!
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
```
Would be great if you could help me out or at least point me in any helpful direction.

Thank you very much in advance!
Harald

basking-in-the-sun2000, 2021-09-06:
Harald,

- I have to push a cleaner display for the first error you showed. It is kind of a minor issue, so I haven't updated the code here. Have a couple of other things that I'm hoping to finish before the next push.

- I think the other error is related to configuration. Did you configure your config.py file?
  - The issue is that you are using unit id 0 in the config file, and the dongle uses id 1. 
  - `slave = 0x00 # seems the dongle uses 0x01 for the slave id`

haralddvorak, 2021-09-08:
Thank you for your fast reply and support!

Exactly the unit id was the problem. This seemingly was a misunderstanding on my part!
I thought the unit id needs to be different from the dongle id.
Now that I have set slave = 0x01 and everything ist working as expected.

Cheers,
Harald

closed 2021-09-08 by haralddvorak