basking-in-the-sun2000/solar-logger #30: read register error - main.py line 99

DavidVM1982, 2021-11-17:
the data of the huawei convert has been working until a few days ago
I get the following error when running main.py

read register error MPPT_N 1
read register error MPPT_N 2
read register error MPPT_N 3
read register error MPPT_N 4
map looping
read register error ESU_charge 1
read register error ESU_charge 2
read register error ESU_charge 3
read register error ESU_charge 4
Wed Nov 17 15:17:58 2021 ESU_charge
map looping
read register error ESU_charge 1
read register error ESU_charge 2
read register error ESU_charge 3
read register error ESU_charge 4
Wed Nov 17 15:18:35 2021 ESU_charge
map looping
read register error M_PTot 1
read register error M_PTot 2
read register error M_PTot 3
read register error M_PTot 4
Wed Nov 17 15:20:26 2021 M_PTot
map looping

DavidVM1982, 2021-11-18:
fix:
in file  modbustcp.py
line 25  removed ", unit=UnitID"

basking-in-the-sun2000, 2021-11-19:
@DavidVM1982 

Glad you solved it. The unit id is a problematic issue, since some inverters use 0 or 1 depending on their ethernet connection. 

Did you mean line 24? The line after data['method'] == "input"? I think there are no registers that use that line, since most are hold type.

This type of error usually points towards a problem with the connection. For instance my inverter is at the edge of wifi coverage. This leads to having to redo calls multiple times. It isn't an abject failure, since you are able to do other calls between those failures.

DavidVM1982, 2021-11-19:
yes,it is line 24    (I add extra  debug lines)

closed 2021-12-27 by basking-in-the-sun2000