basking-in-the-sun2000/solar-logger #32: Solcast 3 direction

verisgit, 2022-04-22:
Hi there, thanks for making this the doco is very detailed and I'm keen to try it out with my SUN2000-5KTL-L1.

I have a couple of initial questions. 
2. My panels face 3 different directions and some are optimised. Is it possible to have 3 rooftop sites for this purpose?
3. Have you tried it with an RPI using a USB RS485? or only using SDongle with Modbus over TCP?

Thanks for the help

verisgit, 2022-04-22:
Seems they now only allow one direction with the free account

basking-in-the-sun2000, 2022-04-22:
Didn't thought of people needing 3 directions, but that should be easy to add in the code. The issue might be if you can get solcast to report 3 directions. Currently it allows you to get 2 different sites (1 direction for each). Let me know if you can get solcast to give you 3 sites (directions), and I can add the option in code.

I had played with the idea of using rs485 directly, since I have 2 meters, and the inverter only reads one. However, I'm not to keen on interconnecting logic and power systems more than needed. Even thought of doing a wireless rs485 link, but tcp modbus is working fine

verisgit, 2022-04-24:
I've contacted them to find out if it's possible will let you know what they say.

basking-in-the-sun2000, 2022-04-24:
thanks. It is a shame they stopped tuning, it gave rather good forecasts

verisgit, 2022-05-09:
Actually, they've suddenly allowed me to have 3 'rooftops' for testing this if there's a way for me to add a 3rd rooftop to test. Obviously, it would need to factor in that the inverter has a limit and given it's aggregating three sites if their output is greater than the inverter limit solcast isn't going to cap it like it does with a single rooftop.

For others who can't get access to additional rooftops in solcast the alternative I was testing with reasonably good results was to have everything facing north then drop the capacity of the additional panels. I think I was working on 30% and it was relatively easy because I had 5 panels in each direction West, East and North. So I treated East and West as operating at 30%. It worked ok for the overall output forcast but not so great for estimating production in the morning or afternoon as it wasn't taking into account the shoulder performance of having east (morning sun) and west (afternoon sun) panels facing directly into the sun.

basking-in-the-sun2000, 2022-05-09:
As the logger stands, it can handle 2. Will be adding a third one in the next few days for your site. You set the configuration with the power capacity for each rooftop (direction) in solcast. In the config file, you specify the capacity for the inverter (forecast_capacity), which will clip the forecast to this value after adding all rooftops. 

Since they stopped tuning, my forecasts has been more off every time. Hopefully, you don't have any shading which would lower the quality of the forecasts.

verisgit, 2022-05-24:
Any joy with that yet?

basking-in-the-sun2000, 2022-05-25:
have been testing some other changes, but those aren't ready. Pushed (https://github.com/basking-in-the-sun2000/solar-logger/commit/b782dea0ff0446f25f9ff50510c66d7975d0de41) the 3 sites. Do you still get 50 daily api calls, or you get more? It Did it as if you only get 50 a day.

verisgit, 2022-05-25:
I believe I only get 10 API calls per day.

basking-in-the-sun2000, 2022-05-25:
In the top of the rooftop page in solcast it states how many calls you have a day. Mine says _50/50 requests per day._ It is set to make less than that many, since it updates the forecast several times during the day

verisgit, 2022-05-26:
Mine only says 10. I'll ask them about that maybe they can increase it?

verisgit, 2022-06-05:
they've increased it to 50

closed 2022-06-06 by basking-in-the-sun2000