basking-in-the-sun2000/solar-logger #33: Issues getting the app up and running

verisgit, 2022-04-22:
Hi there,

Thanks again for your efforts with this project. I've spent some time tonight trying to get it up and running and ran into a number of issues, some I've worked around, but others I can't seem to solve as I'm somewhat new to python. Hopefully, you can point out where I'm going wrong.

I'm using an RPI 4 with Raspberry Pi OS Lite (CLI version). Ethernet to connect to the Pi and then the Pi is connected to the Inverter's WIFI AP so it can access the modbus on port 6607.

Changes I made to setting_up.md in order to get the steps to work:

sudo apt update && sudo apt upgrade
sudo apt install -y gnupg2 curl wget **git** **python3-pip** (required else none of the pip commands work, or the git checkout later on)

Issues I found when trying to startup the app:

1. sudo service influxdb enable - doesn't work. It complains that it's an unrecognised service. But when I check the status it appears it's working so I skipped that
2. When I tried to execute the influxdb user permission steps none of the commands from setting_up.md worked as they required the databases to exist first. So I assumed it would try create it and just tried to start the app.
3. FileNotFoundError: [Errno 2] No such file or directory: 'Huawei.py' - but resolved this by creating one from the template.
4. Finally got it to start but then received these errors initially 
`python3 main.py
mailing not setup
starting
Fri Apr 22 22:48:35 2022
huaweil inv_address error: timed out
huaweil inv_address error: timed out
huaweil inv_address error: timed out
huaweil inv_address error: timed out
huaweil inv_address error: timed out
192.168.200.1
192.168.200.1
main 1 error: list index out of range
main 2 error: list index out of range
top error: database not found: logger_ds
something went wrong
done
starting
Fri Apr 22 22:50:03 2022
huaweil inv_address error: timed out
huaweil inv_address error: timed out
`
5. I've tried the 0x00 and 0x01 options both seem to result in timeouts. I can successfully telnet from the RPI to the inverter on the Modbus port - that seems to work so at least it can see it and the port is open.
6. I tried debug mode but couldn't find any additional info
7. I have no logs in /var/log/solar - so wasn't able to get any additional info there.
8. Config inverter items are as follows
`inverter_ip = "192.168.200.1" # Default ip address in case it can't find it
inverter_port = 6607
slave = 0x00 # seems the dongle uses 0x01 for the slave id (also tried 0x01)
`

Hopefully not too difficult to resolve and I can get it up and running. Keen to play around with it. Thanks in advance for your assistance.

verisgit, 2022-04-22:
I also made a random test to make sure the Pi can actually communicate with the inverter.

Nano modbustest.py
`
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
import time

client = ModbusClient("192.168.200.1", port=6607)
client.connect()

if client.connect():
    time.sleep(1)
    rr = client.read_holding_registers(address=30000, count=15 ,unit=0)
    test = rr.registers[1]
    print(test)
`
python3 modbusttest.py
`20018`

Seems like it's at least reading values correctly. If I check the status it's 0 = Standby which it is right now.

basking-in-the-sun2000, 2022-04-22:
Guess it is a chicken and the egg issue. I might have gotten things running before and tried to write the setup guide after. So some steps might be missing. Also, things have probably changed since. Still need to find a good break to upgrade to influxdb 2, but don't have a dev system for such large changes.

Thanks, added the git and pip. The enable issue might be solved by the installer itself, thus not requiring you to do it. Will leave it in case it doesn't always enable it.

The DB creation is again part of the same chicken salad. The program creates the DB if these haven't been created, but makes it hard to set up things beforehand. lol

The huawei.py file is created by the application. The reason was that I kept changing it, and people might not have been aware of those changes, or they had made changes themselves. Thought it would be better that they added the changes manually after, so they would know. This is also the case for the config.py file.

Maybe Huawei changed things between our inverters, but mine only opens port 502. If you are using rs485 directly, the program isn't set up but for TCP. pymodbus allows for that but needs some changes to the code. Probably very straight-out, but didn't add it since I have no way of testing. Think you would start modbustcp.py with f**rom pymodbus.client.sync import ModbusClient?** 

The first part of opening the Modbus connection is to see if it can find the inverter. That first timeout is that it isn't getting a reply to the broadcast. Wouldn't be too concerned about this error. It tries to figure out the address, in case it has changed. However, Huawei might have changed the way the broadcast works, and might no longer be compatible with your unit.

Made a couple of changes based on your findings, will give it a try before committing.

verisgit, 2022-04-23:
I can confirm I'm still using the Wifi method currently. I do have a serial RS485 connector but I'd probably rather stay with Wifi if I can get it to work for the same reasons you outlined. Also, several people have reported that the WLAN/FE/Sdongle takes around 30 seconds to get the data for each request, whereas Wifi takes around 5 seconds.

Huawei did change the Modbus ports for some of the inverters so it's either 502 or 6607. I confirmed it with a Modbus client on my phone connected to the inverters wifi and 502 is closed, 6607 is open.

I did wonder if maybe it was an issue because it was nighttime, have tried again today and still getting the same results.

Any ideas to try?

basking-in-the-sun2000, 2022-04-23:
You just reminded me of a recent change by Huawei. Not sure if this might be related to your problem

https://forum.huawei.com/enterprise/en/modbus-tcp-guide/thread/789585-100027

verisgit, 2022-04-23:
I don't have an SDongle and I already confirmed I can talk from the RPI to the Inverter over TCP Modbus via manual modbustest.py above. 

For unknown reasons main.py times out, I have no logs, debug mode doesn't provide any debug, etc. Is there something I can do to troubleshoot this?

verisgit, 2022-04-24:
I've done some more troubleshooting. The IP and port are definitely correct as a Modbus client connected to that WIFI network can query the registers and I've confirmed I'm getting valid results, so Modbus TCP is on and working.

Scanner_2.py from your app returns all the register values and appears to be working as expected but that's hardcoded to Unit = 0
Scanner.py runs and outputs several pages of values so I'm assuming that's working fine but not idea what it's supposed to do?
Main.py reports the "huaweil inv_address error: timed out" error and never seems to run. I've reinstalled from scratch on a new RPI image just to make sure and I have the same result. 

Where does Main.py get the Modbus TCP unit id from? My unit has to be 0 or it won't work.

basking-in-the-sun2000, 2022-04-24:
Just pushed a new commit (https://github.com/basking-in-the-sun2000/solar-logger/commit/e99e85997bd9593f22d42a2b98778862b82337af), hopefully this works out for you

verisgit, 2022-04-25:
Thanks, I've merged that into the version I'm tracking and it seems to run now.

`tail -100f /var/log/solar/solar.log
mailing not setup
starting
Mon Apr 25 09:04:32 2022
192.168.200.1
main 1 error: list index out of range
main 2 error: list index out of range
starting
forcast info

0.19047619047619047 0.9082398044844923
`

Still seeing two 'list index out of range' errors? are they important?

verisgit, 2022-04-25:
[image]
[image]

basking-in-the-sun2000, 2022-04-25:
considering it is a new site, guess there are many things that need to be populated. Wait for a day to see if you get any further errors. These are errors that it couldn't grab the data from the db when setting the previous days values. Seems you have a meter, so these should get filled up 

I'm glad the changes worked for you

verisgit, 2022-04-26:
Thanks for your help with this. It's quite cool to see it working now and hopefully I can add some stuff and share it. One thing I have noticed is that the solar forecast appears to be offset to the wrong time.

In the config file example there were quotes are they intended to be there or should they be removed? It's like it's picking up the wrong time zone as my forecast is about 8 hours out.

Should it be

time_zone = 'Country/City'
Or
time_zone = Australia/Perth
Or
time_zone = GMT+X

basking-in-the-sun2000, 2022-04-26:
The config.py should be time_zone = 'Australia/Perth' Guess it is confusing when your country is a continent lol

Also, grafana makes things harder. There should be a tz option for most places. In dashboard > settings > variables, there is also a global variable tzone. Not sure if I pushed the changes over to github. Been tweaking the dashboads and haven't sent these changes in the last commits

Will be closing this issue, since this is a new problem and unrelated to the previous problem

closed 2022-04-26 by basking-in-the-sun2000