basking-in-the-sun2000/solar-logger #39: Timezone issue

verisgit, 2022-04-28:
Hi there,

Thanks for the updates earlier, I've checked the Grafana settings and Grafana itself is set to the correct timezone.

The dashboard variable is set to "Australia/Perth" but I wasn't sure what format it's supposed to be in? Australia/Perth, GMT+8, UTC+08:00, etc.  I tried a few and some caused errors but none moved the solcast forecast to the correct spot.

I also changed the other occurrences in the Dashboard JSON from UTC to Australia/Perth but solcast is still in the same spot. I'm wondering if it's wrong in the database but I assume all of that is stored in UTC so offsets can always be applied to correct it?

The main problem is for some reason the Solcast data is offset by 8hrs. So the charts aren't lined up properly. It's like everything is using the timezone value except for solcast.

[image]
[image]

basking-in-the-sun2000, 2022-04-29:
see if the new commit (https://github.com/basking-in-the-sun2000/solar-logger/commit/bfe6d0f18db7ea4b000525d30c021489b8604dfa) solve your issues

Import the dashboards as described below
https://github.com/basking-in-the-sun2000/solar-logger#grafana-files

verisgit, 2022-04-30:
Thanks trying to set this up now. A few questions regarding units.

tzone - your local timezone (what is the timezone format?)
solcast_adj - if you need to adjust the forecast, 1.0 doesn't change it (what is the unit? is this a factor i.e. 1.2*x)
peak_I - the maximum current the inverter produces (is this in kW? i.e. is a Sun2000-5KTL-L1 5kW Inverter = 5?

verisgit, 2022-04-30:
Here's what the latest dashboard looks like with the changes.
[image]
[image]

basking-in-the-sun2000, 2022-04-30:
> tzone - your local timezone (what is the timezone format?)
Should follow the same format as the one described in the config.py fil
- https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
You could also try Australia/West instead of Australia/Perth Australia/Perth
That is bewildering. All the db calls in grafana have a timezone clause

In the energy panel, go to the edit view and then just below the graph click the **Query inspector** button. The query should look similar to this:

`SELECT cumulative_sum(integral("power90")) * 1.0  /3600   FROM "forcast" WHERE time >= 1651294800000ms and time <= 1651381199999ms GROUP BY time(5m) fill(linear) tz('Australia/Perth');SELECT mean("P_daily") FROM "Huawei" WHERE time >= 1651294800000ms and time <= 1651381199999ms GROUP BY time(5m) fill(none) tz('Australia/Perth');SELECT cumulative_sum(integral("power"))   /3600 * 1.0  FROM "forcast" WHERE time >= 1651294800000ms and time <= 1651381199999ms GROUP BY time(5m) fill(linear) tz('Australia/Perth');SELECT cumulative_sum(integral("power10"))   /3600 * 1.0  FROM "forcast" WHERE time >= 1651294800000ms and time <= 1651381199999ms GROUP BY time(5m) fill(linear) tz('Australia/Perth');SELECT cumulative_sum(integral("M_P")) /3600 / 1000  FROM "Huawei" WHERE time >= 1651294800000ms and time <= 1651381199999ms and "M_A-U" != 0 GROUP BY time(5m)  fill(linear) tz('Australia/Perth');SELECT (integral("power90")) * 1.0  /1800  FROM "forcast" WHERE time >= 1651294800000ms and time <= 1651381199999ms GROUP BY time(5m) fill(linear) tz('Australia/Perth');SELECT mean("P_active")  as prod FROM "Huawei" WHERE time >= 1651294800000ms and time <= 1651381199999ms GROUP BY time(5m) fill(null) tz('Australia/Perth');SELECT (integral("power")) * 1.0  /1800  FROM "forcast" WHERE time >= 1651294800000ms and time <= 1651381199999ms GROUP BY time(5m) fill(linear) tz('Australia/Perth');SELECT (integral("power10")) * 1.0  /1800  FROM "forcast" WHERE time >= 1651294800000ms and time <= 1651381199999ms GROUP BY time(5m) fill(linear) tz('Australia/Perth');SELECT cumulative_sum(integral("M_P")) /3600000 - cumulative_sum(integral("P_active")) /3600  FROM "Huawei" WHERE time >= 1651294800000ms and time <= 1651381199999ms and "M_A-U" != 0 GROUP BY time(5m)  fill(linear) tz('Australia/Perth')`

> solcast_adj - if you need to adjust the forecast, 1.0 doesn't change it (what is the unit? is this a factor i.e. 1.2*x)
It is just a factor that multiplies the solcast forecast for display only 

> peak_I - the maximum current the inverter produces (is this in kW? i.e. is a Sun2000-5KTL-L1 5kW Inverter = 5?
Current in Amperes. Should be in the datasheet for your inverter. The inverter injects current onto the grid, and there is a maximum value. This current can produce work based on the voltage, so your 5kW are really 5kVA. Since you entered 5, you have a line in the power panel around 1.2kW

If your voltage is low you won't reach 5kVA, and if the voltage is high you will surpass this "limit". It uses peak_I to draw a max line in the power panel.

Looking at your inverter, Huawei might be limiting the current to produce a fixed power?!?! Mine only produces 10 kVA at 240V and less at lower voltages. Supposedly the maximum current for yours is 22.7A

verisgit, 2022-05-01:
config.py = time_zone = "Australia/Perth" from https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
I think Huawei limit the inverters to 5kw as that's the max you are allowed to export here.

verisgit, 2022-05-01:
I had a play around with the influx queries in Grafana, weird thing is the "tz" property in Grafana didn't seem to be doing anything even if I made all the datasets different timezones for testing if they would align (or to see if one was wrong).

[image]

this made me wonder what's in the actual database and what could cause them to be offset by this amount. I did find that my RPI was set to UTC time. I have changed it to Australia/Perth. But not sure if that's used anywhere or what the impact would be of suddenly changing it. Hopefully, it doesn't start writing the rest of the data offset.

basking-in-the-sun2000, 2022-05-05:
the timezone clause allows queries to be aligned to your time. This prevents getting multiple values for what should be only one, since you day is probably split among two different days in relation to UTC

Hope your setting the rpi time solved your issues

verisgit, 2022-05-05:
After I set the correct time on the RPI via raspi-config it seems to have fixed it. Thanks for the help with that.

closed 2022-05-05 by verisgit