basking-in-the-sun2000/solar-logger #47: Add new config files for other inverters

andysu, 2022-06-02:
Please add to documentation format Huawei.default.py file
(which values are required for program)
end more info about values in config file.
I have 1 Solis inverter conncted via modbusrtu and 1 ABB
and documentation about registers map.
I can make a config files for them.
I have experience of polling many inverters and I can, 
with documentation, do a poll of any inverter. 
I already did a survey of inverters 
ABB 
Solis 
Kaсo 
Fronius 
PrimeVolt 
in another program

basking-in-the-sun2000, 2022-06-03:
Will be adding some comments on the next push. Will be adding the following
```

# Change use to ext for type you won't use or aren't defined by your inverter
# In case you need to create your own register map, you can use this file as a guide
#
#index is how the logger refers to this value, and will be used to store into the db
#addr is the address of the register to read
#registers is the number of registers to read
#nam is a reaable definition of the register
#scale is a factor to divide the read value by 
#type is used to cast the read value unto a usable value by the logger
#   the options are Bit16, Bit32, I16, I32, str, U16, U32
#   bit16 and bit32 allow for further manipulation of the bits read
#units is a postfix added to the read value to display it correctly
#use is the type of use given to the value
#   the options are data, ext, info, mult, stat
#   data is actual data, which is updated every 30s
#   ext is extra and not used, so you can disable some values by changing the use to ext
#   info is an inverter info value. These are updated when there are changes
#   mult is multiple registers. Same as data, but you don't need to read all, as in the case of PV values
#   stat is a status value. These are updated when there are changes
#method is how the register is read (hold 03 or input 04)

```

andysu, 2022-06-04:
need a list of required registers
those without which it will not work
in principle, monitoring will only work with 1 register
(accumulated energy)

basking-in-the-sun2000, 2022-06-04:
Never thought of it in those term, since I needed all the registers define to work. As these provide me with the info I wanted. However, would need to check main.py for any referenced register. The logger in theory needs the meter, but several people have used it without one. Not sure if the code just gets around, or the protection to prevent a crash is getting around that issue. 

Also expanded the register map you did. Hope it provides most of the values you can need. Hope it works well for you

Check https://github.com/basking-in-the-sun2000/solar-logger/commit/a5b1ddd18f3ea84ccbdcd64856ed51680d7ee111

basking-in-the-sun2000, 2022-06-04:
Registers called by code by name:
Insulation
M_PExp
M_PTot
P_daily
P_peak
PV_In
PV_Un
Shutdown
Start
Temp

These allow for several functions throughout the code, and populate these variables
c_exp
c_gen
c_peak
c_start
c_stop
c_tot
y_gen

types that probably need to have at least one register defined for them
stat, info, data

You would also need to provide these, since they are called by main. Either just with a return or empty
_status_map 
def inv_address():
def status(register, value):
def test_bit(n, b)