basking-in-the-sun2000/solar-logger #48: Daily production stopped

kanagawawave, 2022-06-28:
First thanks for this amazing repo, loving it.
A few days ago, my daily production stopped. I'm doing anything wrong? I got no errors on my /var/log/solar/logger.log

How to proceed?
[image]

basking-in-the-sun2000, 2022-06-28:
set the debug flag in the config.py file to True

You can see the result in /var/log/solar/solar.log. You might get an idea where the application is having trouble

kanagawawave, 2022-06-28:
Yes, the other graphs works fine and my debug is always true, no errors i think just normal logs like:

> Mon Jun 27 22:31:47 2022
Loaded: Huawei
statring setup
opening modbus
192.168.31.13
opening db
setting db
setting params
{'time': '2022-06-26T09:16:09.684665Z', 'Insulation': 3.0}
starting
in forecast
Mon Jun 27 22:31:49 2022
8
20
Mon Jun 27 23:03:07 2022
next forecast Tue Jun 28 07:45:00 2022
loop
read register error Model 1
read register error Model 2
read register error Model 3
read register error Model 4
Mon Jun 27 22:32:15 2022 Model
map looping
read register error Model 1
read register error Model 2
read register error Model 3
read register error Model 4
Mon Jun 27 22:32:41 2022 Model
map looping
read register error Model 1
read register error Model 2
read register error Model 3
read register error Model 4
Mon Jun 27 22:33:06 2022 Model
map looping

basking-in-the-sun2000, 2022-06-28:
Seems you aren't able to poll the inverter. It opens a connection to 192.168.31.13, but not able to read any values. 

If nothing changed (upgrades?), try running scanner.py. This might show if you can poll the inverter

kanagawawave, 2022-06-28:
> Seems you aren't able to poll the inverter. It opens a connection to 192.168.31.13, but not able to read any values.
> 
> If nothing changed (upgrades?), try running scanner.py. This might show if you can poll the inverter

Tried to run scanner and scanner_2 and nothing happens, maybe because night time?
During day running solar.service we got:

> Mon Jun 27 17:58:19 2022
next forecast Mon Jun 27 17:58:19 2022
loop
{'Model': 'SIW300H M040 L1', 'SN': 'XXXX', 'strings': 2, 'Pn': 4.0, 'Pmax': 4.4, 'Smax': 4.4, 'Qmax': 2.64, 'Qgrid': -2.64, 'Insulation': 3.0, 'Start': >21474836.47
21474836.47
21474836.47
21474836.47
14.94
15.96
4.4
2022-06-27 05:56:18
2022-06-26 17:32:54
44.4
3.0
looking at low production
{'PV_P': 0.14, 'U_A-B': 225.2, 'U_A': 203.7, 'U_B': 0.2, 'I_A': 0.647, 'P_peak': 4.4, 'P_active': 0.133, 'P_reactive': 0.001, 'PF': 1.0, 'Frequency': 60.03, 'η'>1656360121.1305065
1656360000
1656360000
0.0898
{'State1': 'grid-connected | grid-connected normally', 'State2': 'unlocked | connected | DSP collecting', 'State3': 'on-grid | off-grid switch disable', 'Alarm1>looking at low production
{'PV_P': 0.133, 'U_A-B': 222.8, 'U_A': 201.4, 'U_B': 0.2, 'I_A': 0.656, 'P_peak': 4.4, 'P_active': 0.126, 'P_reactive': 0.002, 'PF': 1.0, 'Frequency': 60.02, 'η>1656359829.984179
1656360000
1656360000
0.0898

basking-in-the-sun2000, 2022-06-28:
Could be that your inverter goes to sleep at night? Not sure if the hibernating setting would help. Understand some inverters don't provide info at night.

Other than some values off, the log you got during the day seems ok. For instance you efficiency has a weird value. Could be that your WEG inverter ;) might have some registers work differently? You can compare the register map against the Huawei.py which is the default inverter model. However, it was working a week ago.

kanagawawave, 2022-06-28:
Moved from my server to rpi and seems to be working now. I'll keep updated if i find the cause. 
Thanks

closed 2022-06-28 by kanagawawave