basking-in-the-sun2000/solar-logger #50: Question: Backups

verisgit, 2022-08-02:
Hi there, just wondering what files/folders need to be backed up if I want to create a second copy in case my Pi fails. I'd hate to lose my historical data.

Also are there any particular folders to check for max file size or out of limit issues long term?

basking-in-the-sun2000, 2022-08-03:
most is stored in influxdb. You would need to backup your db

verisgit, 2022-08-07:
Where's that located based on the install steps?

Is it just /var/lib/data that I would need to backup or the meta and other folders also?

basking-in-the-sun2000, 2022-08-08:
you shouldn't need to backup the folders. Dbs have a mechanism to backup the data

https://docs.influxdata.com/influxdb/v1.8/administration/backup_and_restore/

verisgit, 2022-08-08:
Ah yes, I've setup a cron to do that. Thanks

closed 2022-08-08 by verisgit