basking-in-the-sun2000/solar-logger #51: Deye Inverter

AtmosphereZA, 2022-10-02:
Anyone have config settings or instructions for setting this up to work with the DEYE inverters?

TIA

basking-in-the-sun2000, 2022-10-02:
The first part would be to create a modbus register map file, similar to Huawei.default.py.  Currently the logger works with modbus tcp, but you should be able to use it with rtu.

The general idea was disused in another [thread](https://github.com/basking-in-the-sun2000/solar-logger/issues/47)

andysu, 2023-12-06:
deye loggers has two ways to communicate
1 ModbusTCP - the usual modbusTCP, but the frames are wrapped in a proprietary protocol. But there are already decryption projects on git (successful)
2 ModbusRTU - original modbus. but registers map has many garbage - remains (imho) from different models. but main registers can be read without problems

sorry for terrible English))