basking-in-the-sun2000/solar-logger #62: Upgrade

verisgit, 2023-09-01:
Hi there, I have an initial version installed from some years ago and haven't managed to keep it updated.

I want to update to the latest version how would you best suggest to do this and be able to preserve my old data?

should I shut it down install a fresh one and go from there? Will the data be compatible? Looking for suggestion on the best path forward. Thank

verisgit, 2023-09-01:
I just gave it a crack earlier. Steps were as follows:

1. Shutdown the services
2. Upgrade RPI
3. Update code (merged all my changes and configs back in)
4. Upgrade the PIP packages for pymodbus, etc. see setup.md
5. Start the services back up and check it's all running ok
6. Update the dashboards
7. Restart - check it still autostarts ok

Post upgrade I'm seeing these errors in the logs. the influx one I believe is new but the modbus one was showing up previously. They happen fairly regularly. @basking-in-the-sun2000 any ideas how to resolve these two:
- modbus read_registers error: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(*.*.*.*:6607)]
- write_influx error: 400: {"error":"unable to parse 'Huawei_daily,location=main 1693497600000000000': invalid field format"}

verisgit, 2023-09-13:
Hi there @basking-in-the-sun2000 I've been noticing some weird issues lately. I seem to have some gaps in data, and also the 365/90 day trackers don't seem to be recording despite there being data for most of the days (there was an actual gap when the system failed, but recently i.e. the last few weeks it's been working).

Any ideas why this would be failing?

[image]
[image]

Also looking like solcast is missing and totals aren't showing up in the dashboard either for yesterday for example
[image]

verisgit, 2023-09-13:
In the logs I'm seeing this:

solar.log
`Wed Sep 13 14:11:30 2023
192.168.200.1
write_influx error: 400: {"error":"unable to parse 'Huawei_daily,location=main 1694534400000000000': invalid field format"}
starting
forecast error: 'forecasts'
solcast off-line
modbus read_registers error: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.200.1:6607)]
done
starting
Wed Sep 13 14:15:30 2023
192.168.200.1
write_influx error: 400: {"error":"unable to parse 'Huawei_daily,location=main 1694534400000000000': invalid field format"}
starting
forecast error: 'forecasts'
solcast off-line
modbus read_registers error: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.200.1:6607)]
done
starting
Wed Sep 13 14:21:16 2023
192.168.200.1
write_influx error: 400: {"error":"unable to parse 'Huawei_daily,location=main 1694534400000000000': invalid field format"}
starting
forecast error: 'forecasts'
solcast off-line
modbus read_registers error: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.200.1:6607)]
done
starting
Wed Sep 13 14:25:03 2023
192.168.200.1
write_influx error: 400: {"error":"unable to parse 'Huawei_daily,location=main 1694534400000000000': invalid field format"}
starting
forecast error: 'forecasts'
solcast off-line
modbus read_registers error: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.200.1:6607)]
done
starting
Wed Sep 13 14:32:31 2023
192.168.200.1
write_influx error: 400: {"error":"unable to parse 'Huawei_daily,location=main 1694534400000000000': invalid field format"}
starting
forecast error: 'forecasts'
solcast off-line
`

verisgit, 2023-10-17:
@basking-in-the-sun2000 are you still working on this project? Haven't seen you surface in a while

aptura-scott, 2023-11-21:
still haven't been able to resolve this, potentially going to try wiping all previous data and see if it works with a fresh database, don't really have any other options.

verisgit, 2023-11-21:
still haven't been able to resolve this, potentially going to try wiping all previous data and see if it works with a fresh database, don't really have any other options.

verisgit, 2023-11-21:
Seems to be failing on main.py
running in debug shows this

`SELECT HOLT_WINTERS_WITH_FIT(first(r), 1, 7)  from (SELECT (P_daily / f_power * f_ratio) as r FROM Huawei_daily WHERE time >=  1700496000000ms -10d and time <= 1700496000000ms tz('***/***')) group by time(1d) tz('***/***')
write_influx error: 400: {"error":"unable to parse 'Huawei_daily,location=main 1700496000000000000': invalid field format"}
error
[{'time': 1700496000000000000, 'measurement': 'Huawei_daily', 'tags': {'location': 'main'}, 'fields': {'f_ratio': None}}]
logger_lt
`

verisgit, 2023-12-15:
It seems the new forecast optimisation features were breaking it; they don't seem to work, and disabling them fixed my issue. I now have reports for each day being generated again.